Mosibi/Midea-heat-pump-ESPHome #1: Error - Decoding

beccsjb, 2023-11-03:
G'day Richard, awesome work and i was super pleased to see that you'd refined it even more since its inception.

Curious, have you done much with the error reporting? It appears that the heatpumps have a vary array of E, H, P & F Fault codes, yet via RS485 it only outputs a single digit. 
I've seen a few random E8's on the Baxi screen and looking at history the RS485 interface is reporting an error code of 9.

Mosibi, 2023-11-03:
Hi,

nope, I did not dive into the error codes yet. It is in the back of my head to work on sometimes. But a pull request is very welcome! ☺️

Mosibi, 2023-11-05:
Hi @beccsjb,

I have just committed an update to heatpump.yaml where the faults (current, 1, 2 and 3) are mapped to an error code. For the current fault, I also created an extra mapping with the error description.

[image]

closed 2023-11-05 by Mosibi