Mosibi/Midea-heat-pump-ESPHome #8: name of values

Yocee84, 2024-01-09:
It would be more useful and more transparent if the entries followed the value names with the manufacturer's serial number.

Because in this way the information is not displayed in logical order on the interface, but based on name

Example:

${entity_prefix} Time Interval Of Compressor Startup In Heating mode"
**228**_${entity_prefix} Time Interval Of Compressor Startup In Heating mode"

Example of listing now for first few values:
Heatpump Ambient Temperature For Enabling Hydraulic Module Auxiliary Electric Heating IBH | -5 °C |  
Heatpump Climate Curve T1S Calculated Value 1 | 255 °C |  
Heatpump Climate Curve T1S Calculated Value 2 | 255 |  
Heatpump Compressor Target Frequency | 0 Hz |  
Heatpump Current Fault | 0 |  
Heatpump DC Bus Current | 0 A |  
Heatpump DC Bus Voltage | 300 V |  
Heatpump Electricity Consumption | 2348 kWh

Example the logical order:
000 Active State Inactive
001 Operational Mode 3
001 Power Air Conditioner Zone 1 Off
001 Power Air Conditioner Zone 2 Off
001 Power Floor Heating Zone 1 Off
001 Power Reserved BIT 10 Off
001 Power Reserved BIT 11 Off
001 Power Reserved BIT 12 Off
001 Power Reserved BIT 13 Off
001 Power Reserved BIT 14 Off
001 Power Reserved BIT 15 Off
001 Power Reserved BIT 4 Off
001 Power Reserved BIT 5 Off
001 Power Reserved BIT 6 Off
001 Power Reserved BIT 7 Off
001 Power Reserved BIT 8 Off
001 Power Reserved BIT 9 Off
003 Heatpump Air Temperature Ts 40 °C
004 DHW Tank Temperature T5s 20 °C
005 Function Setting DHW Pumps Running Constant Temperature Water Recycling Off
005 Function Setting Disinfect Off
005 Function Setting ECO Mode Off
005 Function Setting Holiday Away Off
005 Function Setting Holiday Home Off
005 Function Setting Reserved BIT 0 Off
005 Function Setting Reserved BIT 1 Off
005 Function Setting Reserved BIT 14 Off
005 Function Setting Reserved BIT 15 Off
005 Function Setting Reserved BIT 2 Off
005 Function Setting Reserved BIT 3 Off
005 Function Setting Reserved BIT 9 Off
005 Function Setting Silent Mode Off
005 Function Setting Silent Mode Level Off
007 Forced Water Tank Heating 0
008 Forced Tank Backup Heater 0
009 Forced Hydraulic Module Rear Electric Heater 1 0
010 t_SG_MAX 2 hr

Mosibi, 2024-01-13:
My preference is to keep the numbers out of the name. Sadly the entity ID in HA is based on the 'name' part and not on the 'id' from the esphome config. If that was the case, I probably would have added the registry number and bit number to the ID, since in my opinion that's where it belongs and not in the name.

haute12345, 2024-01-13:
hi, thx for the updates.
perhaps indicating that the heat pump is working, using the fan as an indicator is not always real.
When in defrosting mode, the compressor is running and the fan is off.
Perhaps a more correct way to indicate that the heat pump is working is if it (compressor or fan or pump) is working.

Mosibi, 2024-01-13:
Yeah I don’t remember why I made it so difficult. There is already a “load output run” entity which shows exactly what is needed. Something for the next update…

haute12345, 2024-01-13:
Normally I use "load output run" to indicate compressor operation. But some times I observed that the compressor was running at xHz and this sensor remains off.
What I did was create a threshold sensor in HA to indicate if the Hz is between 1 and 100, the compressor is On.

Mosibi, 2024-01-14:
Ah good pointer! Which exact entity do you use for that?

haute12345, 2024-01-14:
What I did was create a threshold sensor in Home Assistant to indicate if the compressor's Hz is between 1 and 100, the compressor is On. I use a helper, threshold sensor, like bynary_sensor.

Yocee84, 2024-01-14:
> My preference is to keep the numbers out of the name. Sadly the entity ID in HA is based on the 'name' part and not on the 'id' from the esphome config. If that was the case, I probably would have [...]

Thanks for your work!

Yocee84, 2024-01-14:
I have a few more ideas, I'll show you what I use.:

- platform: template
    name: "Heatpump_Average_COP"
    unit_of_measurement: "COP"
    lambda: |-
      return ( (id(${devicename}_power_output).state / id(${devicename}_electricity_consumption).state ) );
    accuracy_decimals: 4
    update_interval: 15s
  - platform: template
    name: "Heatpump_Momentary_COP"
    unit_of_measurement: "COP"
    lambda: |-
      return ( (id(${devicename}_ability_of_hydraulic_module).state / id(${devicename}_outdoor_unit_current).state * id(${devicename}_outdoor_unit_voltage).state /100 ) );
    accuracy_decimals: 4
    update_interval: 15s

Mosibi, 2024-01-14:
> What I did was create a threshold sensor in Home Assistant to indicate if the compressor's Hz is between 1 and 100, the compressor is On. I use a helper, threshold sensor, like bynary_sensor. Yes i use _compressor_operating_frequency for read de Hz.
> 
> But this sensor should only be used to know if the compressor is on. Another thing is to define if the heat pump is on. The heat pump is mainly composed of 3 systems:
> 
>     * Compressor
> 
>     * Fan
> 
>     * Water Pump.
> 
> 
> Since all 3 are not always on. A good way to define if the heat pump is on is to make an OR log with those 3 systems.

Great! See https://github.com/Mosibi/Midea-heat-pump-ESPHome/pull/12

Mosibi, 2024-01-14:
> * Heatpump_Momentary_COP

I also monitor the energy usage of both the internal and internal unit and that learned me that the electricity consumption value(s) are way off. In my case the energy usage that is stated by the heat pump is way less in reality, and that is the reason that I did not include a (s)COP measurement in the configuration.

I am very curious if you also measure the energy consumption in an other way !

haute12345, 2024-01-14:
Yocee84's idea is good, but what you mentioned is true, the internal consumption measurement differs greatly from reality. Measures less current. I use a shelly EM pro meter.
I calculate the COP with a sensor configured in HA similar to the one commented by Yocee84.

Mosibi, 2024-01-14:
> Yocee84's idea is good, but what you mentioned is true, the internal consumption measurement differs greatly from reality. Measures less current. I use a shelly EM pro meter. I calculate the COP with a sensor configured in HA similar to the one commented by Yocee84.

I also use two Shelly's to measure it and do the same in HA. But still I do not trust the numbers completely since I partly must trust the heat pumps metrics. The best way is to use a heat meter

Yocee84, 2024-01-14:
I use a monoblock device.  I also watch with Athom plug (tasmota) and
Sonoff SPM.  The power consumption data are almost identical to the modbus
values.  You have the ohmic heating in the internal device.  This may not
be calculated properly by the machine, especially if it is powered from a
separate phase.  Is the ITB performance specified in the menu?

Richard Arends ***@***.***> ezt írta (időpont: 2024. jan.
14., V 12:57):

Mosibi, 2024-01-14:
> Is the ITB performance specified in the menu?

Where/how can I find/see that?

Yocee84, 2024-01-14:
> > Is the ITB performance specified in the menu?
> 
> Where/how can I find/see that?

For Serviceman - Other Heating source - 7.8 IBH1
250 & 251 register

Mosibi, 2024-01-14:
> > > Is the ITB performance specified in the menu?
> > 
> > 
> > Where/how can I find/see that?
> 
> For Serviceman - Other Heating source - 7.8 IBH1 250 & 251 register

Are both set to "0" on my heat pump

Yocee84, 2024-01-14:
> > > > Is the ITB performance specified in the menu?
> > > 
> > > 
> > > Where/how can I find/see that?
> > 
> > 
> > For Serviceman - Other Heating source - 7.8 IBH1 250 & 251 register
> 
> Are both set to "0" on my heat pump

Press Enter, and setup the correct values. Below 12kw HP is 3kw, Above is 9kw i think.

Mosibi, 2024-01-14:
> > > > > Is the ITB performance specified in the menu?
> > > > 
> > > > 
> > > > Where/how can I find/see that?
> > > 
> > > 
> > > For Serviceman - Other Heating source - 7.8 IBH1 250 & 251 register
> > 
> > 
> > Are both set to "0" on my heat pump
> 
> Press Enter, and setup the correct values. Below 12kw HP is 3kw, Above is 9kw i think.

From the docs: “IBH1 and 2 are the hydraulic module's rear electric heater.”

This week was the first time that IBH 1 was on (heat pump installed for 15 months) and it used 3kW indeed. I changed both to “30”. I can not rationalize why that value is correlated to the energy usage metric of the external unit, but let’s see what happens

Yocee84, 2024-01-14:
I'm looking forward to the numbers!

Another topic:
Wouldn't you like to solve this? :
https://esphome.io/components/climate/thermostat.html

It would also be good for setting the water temperature and changing the mode.

Mosibi, 2024-01-14:
> I'm looking forward to the numbers!
> 
> Another topic: Wouldn't you like to solve this? : https://esphome.io/components/climate/thermostat.html
> 
> It would also be good for setting the water temperature and changing the mode.

:)

That is indeed something that I would like to include. If you have a good idea to implement it, then that is very welcome. A pull request would be great !

Mosibi, 2024-01-14:
> I can't use the pull request part. I can't figure out how to do it.

This is a good explanation: https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github

Basically: fork the repo, create a branch in the forked repo and make changes. Then push the changes to your branch and in the github web gui you will see the option to create a pull request to my repo.

Mosibi, 2024-01-14:
And else, create an issue like this one an place the code in there

Yocee84, 2024-01-14:
> And else, create an issue like this one an place the code in there

Glad to see the new yaml! I start made the thermostat panel.

Yocee84, 2024-01-15:
This adress is maybe number, or select.  

- platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "${entity_prefix} Operational Mode"
    id: "${devicename}_operational_mode"
    register_type: holding
    address: 0x1
    value_type: U_WORD
    
  Select something this:

select:
- platform: modbus_controller
  modbus_controller_id: ${modbus_ctrl_id}
  name: ${friendly_name} Speed
  icon: "mdi:fan"
  id: genvex_sel
  address: 0
  value_type: U_WORD
  optimistic: true
  optionsmap:
    "Low": 1
    "Medium": 2
    "High": 3
    "Turbo": 4
    "Off": 0

Mosibi, 2024-01-15:
This is for the thermostat panel? Tip, place your yaml code between 3 backticks (before and after the code) to format it. Example:

```
example:
  bool: true
  list:
    - one
    - two
```

Yocee84, 2024-01-16:
> This is for the thermostat panel? Tip, place your yaml code between 3 backticks (before and after the code) to format it. Example:
> 
> ```
> example:
>   bool: true
>   list:
>     - one
>     - two
> ```

Not just for thermostat. generally...

Yocee84, 2024-01-16:
I'm a little confused. We can regulate the water temperature in two ways:
- Flow temperature (T1)
- Tank temperature (Tbt1 and/or Tbt2)

Obsession:
- In room thermostat mode based on the air temperature of the KJR-120.

I wonder which should be defined in the case of an integrated room thermostat. I'm betting on T1.

Mosibi, 2024-01-16:
I would start with a climate object for the water tank temperature. So (T5s) to set the tank temperature, T5 (115) as sensor, the option to dis- and enable automatic heating of the tank and the option to force water heating

Yocee84, 2024-01-16:
I think, Select is an prety elegant way:
`select:
- platform: modbus_controller
  modbus_controller_id: ${devicename}
  name: "${friendly_name} Operational Mode Select"
  icon: "mdi:fan"
  id: midea_op_mode_sel
  address: 0x1
  value_type: U_WORD
  optimistic: true
  optionsmap:
    "Heat": 3
    "Cool": 2
    "Auto": 1`

Mosibi, 2024-01-16:
> I think, Select is an prety elegant way: `select:
>
> * platform: modbus_controller
>   modbus_controller_id: ${devicename}
>   name: "${friendly_name} Operational Mode Select"
>   icon: "mdi:fan"
>   id: [...]

Being able to select the mode is indeed very nice. Did you test this code?

Yocee84, 2024-01-16:
> Being able to select the mode is indeed very nice. Did you test this code?

Yes, I tested, and working.

Mosibi, 2024-01-16:
> Yes, I tested, and working.

Cool. This weekend I will test your code and implement it when I am done. Thanks so far ! 👍

Mosibi, 2024-01-17:
> Need convert to swich the next adresses: 210 (PLC: 40211) (Nice for setup new heatpump system) 211(PLC:40212) (Nice for setup new heatpump system)
> 
> Need convert to select: 269 (PLC: 40270) (Nice if the Current or cable is limited)
> 
> Need convert to Number: 270 (P LC: 40271) 271 (PLC: 40272) 272 (PLC: 40273)

The above is already on my todo. 270, 271 and 272 are a bit more work the regular registers, that’s why I did not convert them this weekend when I did the others.

Again, if you have working code, then that would make me very happy 😋

Mosibi, 2024-01-17:
I will see what I can do this weekend. I hope that I have the time to dive into it.

Yocee84, 2024-01-18:
Numbers need define range.
235 Delay Time Of Enabling (15-120 min)
237 Ambient Temperature Trigger (-15 - +10 c)
205 - 208 Temp limits

Yocee84, 2024-01-24:
Some Registers need "Unit of measurement", and "Device class" .

# Register: 120
  # Midea: Tbt1
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "${entity_prefix} Hydraulic Module Current 1"
    id: "${devicename}_hydraulic_module_current_1"
    register_type: holding
    address: 0x78
    value_type: U_WORD
    **unit_of_measurement: "°C"
    device_class: "temperature"**

Mosibi, 2024-02-11:
@Yocee84 I just released version 3.0.0 with a lot of changes. I am closing this issue now, I think all the things discussed are now included. Please open a new one (for each issue) when you find something. 

Thanks for your input !

closed 2024-02-11 by Mosibi