Mosibi/Midea-heat-pump-ESPHome #17: Registers 205 and 206 give an error when used as number

Mosibi, 2024-01-20:
```
[13:53:01][D][number:012]: ' Temperature Lower Limit Of TS Setting': Sending state 17.000000
[13:53:03][D][number:054]: ' Temperature Lower Limit Of TS Setting' - Setting number value
```
Setting the value to 16 results in an error. Function 0x06 is "Write Single Holding Register", exception 1 is "Illegal Function, Function code received in the query is not recognized or allowed by server"

```
[13:53:03][D][number:113]:   New number value: 16.000000
[13:53:03][D][modbus.number:061]: Updating register: connected Sensor= Temperature Lower Limit Of TS Setting start address=0xCE register count=1 new value=16.00 (val=32.00)
[13:53:03][D][number:012]: ' Temperature Lower Limit Of TS Setting': Sending state 16.000000
[13:53:03][D][modbus:119]: Modbus error function code: 0x86 exception: 1
[13:53:03][E][modbus_controller:091]: Modbus error function code: 0x6 exception: 1 
[13:53:03][E][modbus_controller:100]: Modbus error - last command: function code=0x6  register address = 0xCE  registers count=1 payload size=2
```

And the next read show the original value again

```
[13:53:10][D][modbus.number:023]: Number new state : 17.00
[13:53:11][D][number:012]: ' Temperature Lower Limit Of TS Setting': Sending state 17.000000
```

Mosibi, 2024-01-20:
Tried to use lamba code on register 205 and the more easy `multiply`, both results in the same error

```
 # Register: 205, default: 60, TODO: verify default and min/max value
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "${entity_prefix} Temperature Upper Limit Of TS Setting"
    id: "${devicename}_temperature_upper_limit_of_ts_setting"
    register_type: holding
    address: 0xcd
    value_type: U_WORD
    unit_of_measurement: "°C"
    entity_category: config
    lambda: |-
      int value = x * 0.5;
      ESP_LOGI("main", "Sensor value reg 205: %f", x);
      ESP_LOGI("main", "Display value reg 205: %d", value);
      return value;
    write_lambda: |-
      int value = x * 2;
      ESP_LOGI("main", "Incoming value to write for reg 205: %f", x);
      ESP_LOGI("main", "Write value reg 205: %d", value);

      esphome::modbus_controller::ModbusCommandItem set_payload_command = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(${devicename}, 0xcd, value);
      ${devicename}->queue_command(set_payload_command);

      return {};
    min_value: 0
    max_value: 60
  # Register: 206, default: 34, TODO: verify default and min/max value
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "${entity_prefix} Temperature Lower Limit Of TS Setting"
    id: "${devicename}_temperature_lower_limit_of_ts_setting"
    register_type: holding
    address: 0xce
    value_type: U_WORD
    unit_of_measurement: "°C"
    entity_category: config
    multiply: 2
    min_value: 0
    max_value: 60
```

Yocee84, 2024-01-20:
Register addresses 200-208 can only use the 03 (register read only) function code. Register addresses 209 and following can use 03H, 06H (single register write) or 10H (multiple register write)

Mosibi, 2024-01-20:
Hmm, then I have all registers in that range to be a sensor. How did you find that info btw, I tried to find it !

Mosibi, 2024-01-20:
Ah now with your text I could find where it came from. In the manual of course 😄

closed 2024-01-20 by Mosibi