Mosibi/Midea-heat-pump-ESPHome #22: Error modbus

alexswart, 2024-01-26:
I dont get the data good to homeassistant and get the following errors, dont knownwhere i have to look to resolve it
`
10:06:12][E][modbus_controller:091]: Modbus error function code: 0x3 exception: 2 
[10:06:12][E][modbus_controller:100]: Modbus error - last command: function code=0x3  register address = 0x64  registers count=47 payload size=0
[10:06:13][W][component:214]: Component modbus_controller took a long time for an operation (0.06 s).
[10:06:13][W][component:215]: Components should block for at most 20-30ms.
[10:06:13][E][modbus_controller:091]: Modbus error function code: 0x3 exception: 2 
[10:06:13][E][modbus_controller:100]: Modbus error - last command: function code=0x3  register address = 0xF0  registers count=33 payload size=0
[10:06:22][E][modbus_controller:091]: Modbus error function code: 0x3 exception: 2 
[10:06:22][E][modbus_controller:100]: Modbus error - last command: function code=0x3  register address = 0x64  registers count=47 payload size=0
[10:06:22][W][component:214]: Component modbus_controller took a long time for an operation (0.06 s).
[10:06:22][W][component:215]: Components should block for at most 20-30ms.
`

Mosibi, 2024-01-26:
Which exact heatpump do you have? If you have an external and internal unit, both model versions are important.

alexswart, 2024-01-26:
It is an ferroli omnia hybride with an 6kw outdoor unit R410
When i see the register adresses they are all the same for all the midea units.
Some functions are working, i can change the outlet water temperaturen setpoint

Mosibi, 2024-01-27:
I can’t find the modbus overview for a hybrid model at the moment. The model you gave is to broad btw, the exact model, for example “ferroli omnia s hybrid c 3.2” is important to find the right info.

But in general, it seems that register 100 (0x64) is not supported on your device. Remove that one from the list and then try again. Hopefully that is the only register that’s missing.

Please let me know the outcome, including your exact (!) model, so that we can help others as well.

alexswart, 2024-01-27:
That is strenge, i have check my old integration rs485-> usb-> nonered and that works also with adress 100 and shows the frequency of the compressor, i want to check my serial-> 485 board maybe that is the problem

buitre-buitaker, 2024-03-23:
Hi @alexswart , I am having similar problems with a Kosner unit (Midea clone), and I have a rs485 --> serial TTL converter. Did you find any solution for the "Component modbus_controller took a long time for an operation " messages?

Thank you!

alexswart, 2024-03-25:
@buitre-buitaker still no solution for this, i have also try diferent esp board en diferent ttl converters but all give the same error

Mosibi, 2024-03-25:
```
Component modbus_controller took a long time for an operation
```

This message is an informational message from ESPHome and not an error message. It prefers to handle operations within the mentioned time, but that's unfortunately not always possible.

ebbenberg, 2024-05-30:
You can filter out the component message by using this code

```
# Enable/Disable logging
logger:
  logs:
    modbus_controller.sensor: DEBUG
    modbus_controller.output: DEBUG
    modbus.number: DEBUG
    esp32.preferences: DEBUG
    sensor: DEBUG
    text_sensor: DEBUG
    button: DEBUG
    number: DEBUG
    component: ERROR
```

closed 2024-08-05 by Mosibi