Mosibi/Midea-heat-pump-ESPHome #23: Request for Advice

BettySwallocks, 2024-01-26:
A request for advice rather than an actual issue...

I'm using a Midea M-Thermal ASHP (air-to-water) system in the UK with a separate Midea M-Thermal control unit.  I'd like to read data from that system into Home Assistant via ESPHome via an ESP32 Devkit board.  I came across your project and wondered if I could use or adapt it for my needs.  Only looking to read & display the Modbus data right now as I already have a management system,

After checking that you have a custom board available, but that it has no stock I thought I would load your yaml onto my ESP32 WROOM devkitC and see if I could get any data.  Using pins 1 and 3 for TX/RX I connected into H1 and H2 on the controller,  I then watched to see what comes in, but it seems that regardless I don't get anything useful, mostly command timeouts and CRC check failures. Swapping wires doesn't help and it's even questionable if I'm actually getting ANY real data at all as I've got similar output just running it on my desk unconnected to the Midea controller.

I have tried changing pins used, and also the modbus address from 0x0, 0x1, 0x16 etc.  While I have some coding experience and a little prior use of ESP32 and other micro-controllers I'm totally new to Modbus and have no understanding of UART.

My real questions are:
Can I even use a plain ESP32 to read in data using your config?
Why are there only 2 connections at the controller board (H1/H1) but your yaml references TX, RX and a Modbus pin?
Am I completely misunderstanding the connections into the MCU or the way this functions?

Hope you can find time to help.

timmccor, 2024-01-26:
I've got the same Heatpump as you, and I also just wired in an existing ESP32, mainly because they don't ship the premade stuff to the UK.

You've most likely got a wiring problem. Can you show how you wired the ESP32 to the RS485 controller, and how you wired that to the heatpump controller?

BettySwallocks, 2024-01-26:
I think you've already started to answer my question, so thanks.

As a newbie to Modbus, UART I didn't see that I needed to also involve an RS485 controller board.

I'm guessing something like this? : https://www.aliexpress.com/item/1005001621798947.html

timmccor, 2024-01-26:
Ah, yes. The board this project sells is a custom job which is an ESP32 & RS485 controller on the same PCB.

Any should work, I bought this one from Amazon because it was quicker: https://www.amazon.co.uk/gp/product/B07B667STP/

BettySwallocks, 2024-01-26:
Great thanks again, I'll get somehing ordered. Any advice then on connections from new RS485 to ESP32 board as they will be separate?  I'll probably also order a couple of larger breadboards anyway to help.

timmccor, 2024-01-26:
The Amazon one comes with a cable which you can use to easily connect the two boards together via the headers. I used this to help with the pinout: https://embeddedthere.com/how-to-interface-esp32-with-rs485-modbus-sensors-with-example-code/

If you can't get it working let me know and I'll help.

Just make sure you feed each board the correct voltage.

Mosibi, 2024-01-26:
> I've got the same Heatpump as you, and I also just wired in an existing ESP32, mainly because they don't ship the premade stuff to the UK.

We do ship to the UK, we’ve already done that a few times.

Mosibi, 2024-01-26:
I think we have the same setup as you described, with an ext- and internal unit. My advise is to do a good check on the power level and if it’s AC or DC. It’s probably AC, but newer units can use DC. Our solution handles both, so if you are into electronics, it should nog be a problem!

Mosibi, 2024-01-26:
Closing the issue, since it seems to be solved

closed 2024-01-26 by Mosibi

timmccor, 2024-01-27:
> > I've got the same Heatpump as you, and I also just wired in an existing ESP32, mainly because they don't ship the premade stuff to the UK.
> 
> We do ship to the UK, we’ve already done that a few times.

Apologies for the misinformation @Mosibi, I would have sworn you didn't as I looked at ordering one ages ago! I may still as I've not yet gotten around to making mine look pretty.

Mosibi, 2024-01-27:
> Apologies for the misinformation @Mosibi, I would have sworn you didn't as I looked at ordering one ages ago! I may still as I've not yet gotten around to making mine look pretty.

In the very first beginning the UK was indeed not in our "countries-we-ship-to" list, due to very high shipping costs, but luckily we do now for a while with very decent shipping rates

BettySwallocks, 2024-02-04:
I realise this is a closed issue but I wanted to say I have a basic Midea M-Thermal Heat Pump monitoring solution running now, thanks to @Mosibi and @timmccor for effort and advice.Now I have to tune out noise and figure out how to also drop the occasional out-of-range or false temp readings that sometimes come in.