Mosibi/Midea-heat-pump-ESPHome #28: No response received - removed from send queue

bobhacs, 2024-04-23:
Hi

Am trying to use this with a Riello NXHM which I believe is another Midea rebadge. There aren’t any Riello published modbus registers, but Riello just sent me a copy of the Midea ones. 

I’m not getting any data through into HA, except from very rare oddities (e.g. a temperature in the 1000s). 

This is an example error in the log - 

[09:54:36][D][modbus_controller:043]: Modbus command to device=1 register=0x65 countdown=0 no response received - removed from send queue

There are also lots of CRC check failed errors. 

I have a Homely Energy controller connected to the heat pump controller, which uses Modbus. Not sure if that’s part of the issue (but when that’s powered down I still don’t get any data).

Any clues? Using ESP32 with an RS485 TTL adapter. 

Thanks.

Mosibi, 2024-04-23:
Assuming you are using the right connections, basically the same as the Homely does, the first thing I can think of is that you maybe do not have a 120 ohm resistor between the modbus connections

bobhacs, 2024-04-30:
Thanks. Resistor was in place, but it turned out to be a faulty RS485 TTL board, sigh. Tried the third one from the pack and it instantly worked. The ESPhome code is great, thank you. Have found the need to use median and/or clamp filters to get rid of spikes.

closed 2024-04-30 by bobhacs