Mosibi/Midea-heat-pump-ESPHome #35: Status unknown

bkbartk, 2024-10-01:
Hello,

At the moment I get "Heatpump status" "Unknown".
I checked the code for the sensor and this is like this:
```
lambda: |-
      if (id(${devicename}_load_output_run).state) {
        // The heat pump is on
        if (id(${devicename}_status_bit_1_defrosting).state) {
          return {"Defrosting"};
        } else if (id(${devicename}_load_output_sv1).state) {
          return {"DHW"};
        } else {
          if (id(${devicename}_status_bit_1_heating_mode_set_by_room_thermostat).state) {
            return {"Heating"};
          } else if (id(${devicename}_status_bit_1_cooling_mode_set_by_room_thermostat).state) {
            return {"Cooling"};
          } else {
            return {"Unknown"};
          }
        }
      } else {
        // The heat pump is off
        return {"Inactive"};
      }
```

But I use this swich for enabling the heatpump
`switch.heatpump_power_floor_heating_zone_1` and disabled the room thermostat.
would it make sense to add an `or` statement here? and maybe also for cooling?

I can check If I'm able to do somthing here, however I have a pending pull request and I don't know how to submint multiple pending pullrequests to the same project.

Next to that I noticed `# Register: 0 -> Bit 3` is `Power Air Conditioner Zone 2` but in my case the documentation says `Power Floor Heating Zone 2` I don't know if this is device specific, I noticed another PR switching heat and air conditioning.

Mosibi, 2024-10-01:
The 'or' should work indeed, I would love to see a pull request with your changes for this.  Regarding the naming, I used the naming as I've them in the manual for our heatpump and I see that most midea/clone manuals lot use the 'air conditioner'  naming.

If you open a pull request, that triggers me to re-check the manuals ;)

gisbertg, 2024-10-01:
I have 2 manuals here. The original one for my heat pump, a Midea MHC-V8W and a German one for a Remeha Tensio c. Both contain the same information  for register 0 as  @bkbartk  's manual.

rysiulg, 2024-10-03:
My implementation ot statuses in text_sensor: -i've also had problems with that ;)
  # Active State
  - platform: template
    name: "${entity_prefix} Active State"
    id: "${id_entity_prefix}active_state"
    entity_category: "diagnostic"
    icon: mdi:power
    lambda: |-
      std::string operating_mode = id(${id_entity_prefix}operating_mode).state;
      if (id(${id_entity_prefix}load_output_run).state or id(${id_entity_prefix}load_output_water_pump_pump_i).state) {
        // The heat pump is on
        if (id(${id_entity_prefix}status_bit_1_defrosting).state) {
          return {"Defrosting"};
        } else if (id(${id_entity_prefix}load_output_sv1).state and operating_mode == "DHW Heating") {
          return {"DHW"};
        } else {
          if (id(${id_entity_prefix}status_bit_1_heating_mode_set_by_room_thermostat).state or operating_mode == "CO Heating") {
            return {"Heating"};
          } else if (id(${id_entity_prefix}status_bit_1_cooling_mode_set_by_room_thermostat).state or operating_mode == "Cooling") {
            return {"Cooling"};
          } else {
            return {"Unknown"};
          }
        }
      } else {
        // The heat pump is off
        return {"Inactive"};
      }

and   

  # Register: 101
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "${entity_prefix} Operating Mode"
    id: "${id_entity_prefix}operating_mode"
    icon: mdi:state-machine
    register_type: holding
    response_size: 2
    address: 101
    lambda: |-
        int idx = item->offset;
        uint16_t rawdata = (uint16_t(data[idx]) << 8) + uint16_t(data[idx + 1]);
        std::string output = String(rawdata).c_str();
        ESP_LOGI("OPERATING MODE","STan pracy %d", rawdata);
        if (output == "0") output = "OFF";
        else if (output == "2") output = "Cooling";
        else if (output == "3") output = "CO Heating";
        else if (output == "5") output = "DHW Heating";
        return output;
#    filters: 
#      map:
#        0 -> "OFF"
#        2 -> "Cooling"
#        3 -> "CO Heating"
#        5 -> "DHW Heating"

Mosibi, 2024-10-04:
@bkbartk / @gisbertg I rechecked the manuals and the conclusion is that of course there are not consistent at all :)

However the manuals for the latest models are not mentioning 'floor heating' or 'air conditioning' anymore. They state it as:

- Bit 0: power off Zone 1/2; 1: power on Zone 1/2; (room temperature control)
- Bit 1: power off Zone 1; 1: power on Zone 1; (water flow temperature control)
- Bit 3: power off Zone 2; 1: power on Zone 2; (water flow temperature control)

And this makes a bit more sense to me, so I am thinking about changing those registers to:

- Bit 0: Room temperature control
- Bit 1: Water flow temperature control zone 1
- Bit 3: Water flow temperature control zone 2

Any input/remarks on this?

bkbartk, 2024-10-04:
I think this makes more sense, however
I haven't used cooling yet, but I expected Bit 0 needed to be set for cooling,
If the new naming convention is correct then how should I toggle between heating and cooling?

bkbartk, 2024-10-04:
ah, clear,
I assumed, because this one also has the "Auto" option those are the possible modes, not the active one,
But if this is the case, renaming them to the proposed names would prevent others from making the same assumption.

Now I'm wondering what the "auto" mode does, and if I want to heat If I need to set Bit 0,1 of 3?
Now I only set Bit 1 which works.

But maybe one of the others work better.

bkbartk, 2024-10-06:
With the new naming, which is clear now for me,
I think the state should be something like this

```
  - platform: template
    name: "${entity_prefix} Active State"
    id: "${devicename}_active_state"
    lambda: |-
      if (id(${devicename}_load_output_run).state) {
        // The heat pump is on
        if (id(${devicename}_status_bit_1_defrosting).state) {
          return {"Defrosting"};
        } else if (id(${devicename}_load_output_sv1).state) {
          return {"DHW"};
        } else {
          switch(id(${devicename}_operating_mode).state) {
            case 0:
              return {"Inactive"};
              break;
            case 2:
              return {"Cooling"};
              break;
            case 3:
              return {"Heating"};
              break;
            case 5:
              return {"DHW"};
              break;
            default:
              return {"Unknown"};
          }
          return {"Unknown"};
        }
      } else {
        // The heat pump is off
        return {"Inactive"};
      }
```
because we cannot rely on the switches.
Case 0 and 5 probably should never be hit, but just in case

Mosibi, 2024-10-07:
“operating_mode” is the sensor for the selector “operational mode” and does not tell what the heatpump is doing at the moment. See the included picture. (warmtepomp is Dutch for heatpump)

[image]

Mosibi, 2024-10-12:
This issue is solved. If anything new pops up, then please open a new issue.

Again, thanks for submitting and discussing this !

closed 2024-10-12 by Mosibi

bkbartk, 2024-10-12:
thnx, for now I leave it as is because I can't think off a better way.
still "_water_flow_temperature_control_zone_1" indicates heating or cooling. but does not specify which one.

Maybe when I have to time(don't expect it to be soon) I will dive into the documentation to look for a better way, but for now we can leave it closed.

rysiulg, 2024-10-19:
auto mode change from heat to cool and reverse dependent on temperature i think t4 outside plus some temp registers to determine min max for heat and cool ;)PozdrawiamRyszard GwizdakMARM.pl Sp. z o.o.