Mosibi/Midea-heat-pump-ESPHome #40: Midea Wired Remote Controller KJR-120

wizardmelon, 2024-10-04:
Hello,

I’m planning to buy a Ferroli Heat Pump that uses a remote wired controller based on the Midea KJR-120 Modbus mapping.

After comparing the mapping from the file provided with the remote control to the addresses in the YAML file published here, many of the addresses don’t seem to match. Unfortunately, I’m not very familiar with RS485 protocols, and I was hoping someone could help me understand how to modify the YAML file to make everything work.

I don’t have the heat pump installed yet, so I’m unable to run tests at the moment, but I want to start writing the ESPHome code and have it ready for when the heat pump gets installed.

Any help would be greatly appreciated!

Thanks

[
[120F-modbus-mapping-table-0052003366314-V-B.pdf](https://github.com/user-attachments/files/17251817/120F-modbus-mapping-table-0052003366314-V-B.pdf)
](url)

bkbartk, 2024-10-04:
Makes me wonder,
I see a difference for the registers 11 till 20,
(mine has 11,12 and 13, which are not in, but I don't miss them)
But is almost looks like 14 till 17 are the same as register 0 bit 0 to 3
also registers 187 till 197 seems to be new, and everything above 272

rysiulg, 2024-10-22:
@wizardmelon Using this interface -your heatpump will be in 80-90% working with this implementation. 
In your heatpump registry 148-187 is more statistical view in real time and divided COP for DHW/ heating etc

CJOWood, 2024-11-16:
Curious if you got anywhere with this. I just had a ducted Heat Pump installed (rebrand of Midea EVOX) with a KJR-120N controller. Unfortunately, there are no posted modbus register lists that I can find.

I am hoping I can use some trial-and-error to get some basic information and controls.

Mosibi, 2024-12-13:
I think the original question is answered, closing this one

closed 2024-12-13 by Mosibi