Mosibi/Midea-heat-pump-ESPHome #50: Register 103 PMV Openness

rysiulg, 2024-10-18:
HI
My York heatpump have value range from 0 to 480 -so I add filter to get value in % -can you confirm that in your HeatPump (Midea) is similar range? If yes we can update to get this value in percent ;)
Code changes this move in range 0-480 to 0-100%:
```c++
  # Register: 103
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "PMV Openness"
    id: "${devicename}_pmv_openness"
    icon: mdi:valve
    register_type: holding
    address: 0x67
    value_type: U_WORD
    unit_of_measurement: "%"
    filters: 
    - calibrate_linear:  #York have movement 0-480
      - 0 -> 0.0
      - 480 -> 100.0
```

Mosibi, 2024-10-19:
Good one, I see the same on our heat pump indeed. We also have a Midea clone btw, an Airwell.

gisbertg, 2024-10-20:
The same one on my original Midea...

closed 2024-10-22 by rysiulg