Mosibi/Midea-heat-pump-ESPHome #56: Register 272

rysiulg, 2024-10-21:
Hi
Very nice mapping... in lambda :
#
```c++
  # Register: 272 (bits 0-3), default: 1
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "Zone 1 End Heating Mode Emission Type"
    id: "${devicename}_zone_1_end_heating_mode_emission_type"
    icon: mdi:heat-wave
    address: 0x110
    value_type: U_WORD
    optimistic: true
    optionsmap:
      "Fan Coil Unit": 0
      "Radiator": 1
      "Underfloor Heating": 2
    lambda: |-
      // Update the global var unmasked_value_register_272
      id(unmasked_value_register_272) = x;

      // ESP_LOGI("","unmasked_value_register_272: %d",id(unmasked_value_register_272));

      // Extracting 4-bit values
      uint8_t zone2_c_emission = (x >> 12) & 0xF;
      uint8_t zone1_c_emission = (x >> 8) & 0xF;
      uint8_t zone2_h_emission = (x >> 4) & 0xF;
      uint8_t zone1_h_emission = x & 0xF;

      // ESP_LOGI("","zone2_c_emission is %d", zone2_c_emission);
      // ESP_LOGI("","zone1_c_emission is %d", zone1_c_emission);
      // ESP_LOGI("","zone2_h_emission is %d", zone2_h_emission);
      // ESP_LOGI("","zone1_h_emission is %d", zone1_h_emission);

      // Map/dictionary with the possible values
      std::map<int, std::string> emission_options;
      emission_options[0] = "Fan Coil Unit";
      emission_options[1] = "Radiator";
      emission_options[2] = "Underfloor Heating";

      return emission_options[zone1_h_emission];
    write_lambda: |-
      ESP_LOGI("zone1_h_emission", "Set option to %s (%lld)", x.c_str(), value);

      uint16_t new_value = id(unmasked_value_register_272);  // The original unmasked value
      uint8_t value_byte = value;                            // New byte value with the heating emission type for zone 1

      new_value &= 0xFFF0;     // Clear the first 4 bits
      new_value |= value_byte; // Set the first 4 bits with the new value

      // ESP_LOGI("main", "Original value: %d", id(unmasked_value_register_272));
      // ESP_LOGI("main", "New value: %d", new_value);

      esphome::modbus_controller::ModbusCommandItem set_payload_command = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(${devicename}, 0x110, new_value);
      ${devicename}->queue_command(set_payload_command);

      return {};
``` 
but I see possible out of range array error -if from some reason there will be 4th option (based on 2bits register value whe have 4 options) than it generate error.
i suggest to add eg: emission_options[3] = "Unknown";
or better change it to if as example below (i also shorten lines in write_lambda):

```c++
  # Register: 272 (bits 0-3), default: 1
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "Zone 1 End Heating Mode Emission Type"
    id: "${devicename}_zone_1_end_heating_mode_emission_type"
    icon: mdi:heat-wave
    address: 0x110
    value_type: U_WORD
    optimistic: true
    optionsmap:
      "Fan Coil Unit": 0
      "Radiator": 1
      "Underfloor Heating": 2
    lambda: |-
      // Update the global var unmasked_value_register_272
      id(unmasked_value_register_272) = x;
      // ESP_LOGI("Register 272","unmasked_value_register_272: %d",id(unmasked_value_register_272));
      // Extracting 4-bit values
      uint8_t zone1_h_emission = x & 0xF;
      // ESP_LOGI("Register 272","zone1_h_emission is %d"zone2_h_emission);
      // Map/dictionary with the possible values
      std::string string_result;
      if (zone1_h_emission == 0) {
        string_result = "Fan Coil Unit";
      } else if (zone1_h_emission == 1) {
        string_result = "Radiator";
      } else if (zone1_h_emission == 2) {
        string_result = "Underfloor Heating";
      } else {
        string_result = "Unknown: ";
        string_result += std::to_string(zone1_h_emission);
      }
      return string_result;
    write_lambda: |-
      ESP_LOGI("Register 272", "zone1_h_emission Set option to %s (%lld)", x.c_str(), value);
      uint16_t new_value = id(unmasked_value_register_272);  // The original unmasked value
      uint8_t value_byte = value;                            // New byte value with the heating emission type for zone 1
      new_value &= 0xFFF0;     // Clear the first 4 bits
      new_value |= value_byte; // Set the first 4 bits with the new value
      // ESP_LOGD("Register 272", "Original value: %d", id(unmasked_value_register_272));
      // ESP_LOGD("Register 272", "New value: %d", new_value);
      esphome::modbus_controller::ModbusCommandItem set_payload_command = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(${devicename}, 0x110, new_value);
      ${devicename}->queue_command(set_payload_command);
      return {};
``` 

I think is better to consult here before make pull request ;)

Mosibi, 2024-10-21:
I am a bit double in this. The information is read from the heatpump, if it sends something else, then the heatpump is out of spec.

if we really want to handle this, then I would test if the map returns something an set it to “Unknown” if the returned string is empty. But I do not know if c++ is able to handle this in a nice way

rysiulg, 2024-10-21:
OK. For now it is possible to work ok, but if for any reason eg change medea in firmware that extend values than I'm 100% sure that make esp to reset as result out of range array if input (readed from heatpump and process value in lambda) value is above 2 ;)
In this case we have 2 bits per sensor which have 4 possible combinations. So to make sure to avoid this type of error is define array with unknown value for this case or replace by "if" checks which is more trusted. Probably this don't happen, but good is check possibly leaks ;) -like test value before dividing -to avoid division by 0 ;) -that is only my suggestion to disguss -thats why i don't create pull request for that ;)

#The information is read from the heatpump, if it sends something else, then the heatpump is out of spec.
Yes it will be out of spec but when you send value to heatpump, but when you read it then lambda part of code with higher value than 2 makes out-of-range array error and make esp to reboot

Mosibi, 2024-10-21:
You convinced me 😉 Go ahead and make a pull request for it

rysiulg, 2024-10-21:
I've make mistake -there is 4 bits -so 16 combinations ;) -I've looked in programistic calculator ;) looking on modbus specification bits 0-3, bits 4-7, bits 8-11, bits 12-15 -so every part have 4 registers. ;)

I was wondering about 101 register -which version shows you:
Heatpump Controller Software Version  sensor.heatpump_software_version    -in my pump is 24
Wired Controller Version Number    sensor.heatpump_wired_controller_version_number   -in my pump is 27

Mosibi, 2024-10-27:
My heatpump returns 99 for the software version and the wired controller version is 74

closed 2024-11-07 by rysiulg