Mosibi/Midea-heat-pump-ESPHome #57: Register 7, 8, 9

rysiulg, 2024-10-21:
Hi
In SM these registers are configurable. I don't have any heater to force it -maybe something must be also set in config to be working ;)
Can you try look what happens when you "force DHW heating" or "fast DHW" in Smart Home mobile app -this option enables electrical heater in DHW and it changes Register 7 value from 0 to 1, but in my case after few seconds it is disabled.
In Modbus register description in York there is info that 0 is invalid, 1 = Forced On, and 2 = Forced Off... but from mobile Smart Home App values are 0 or 1 and never have 2.
If moved this (and also register 8 and 9) to switch than the code looks like this (and we can ignore off value in this check except write_lambda -but it can be assumed as 0 means off not value 2):
```c++
  # Register: 7
  - platform: modbus_controller
    modbus_controller_id: "${devicename}"
    name: "Forced Water Tank Heater On/Off"
    id: "${devicename}_forced_water_tank_heating_on_off"
    icon: mdi:fire-alert
    address: 0x7
    register_type: holding
    entity_category: config
    lambda: "return x == 1;"
    on_turn_on: 
      then:
        - lambda: |-
            uint16_t value = 1;
            ESP_LOGI("Register 7", "Set forced water tank Heater ON");
            esphome::modbus_controller::ModbusCommandItem set_payload_command = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(${devicename}, 0x7, value);
            ${devicename}->queue_command(set_payload_command);
    on_turn_off:
      then:
        - lambda: |-
            uint16_t value = 0;     // Documentation says 2 but values from phone app makes 0 (according to documentation its invalid, but documentation can be wrong...)
            ESP_LOGI("Register 7", "Set forced water tank Heater OFF");
            esphome::modbus_controller::ModbusCommandItem set_payload_command = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(${devicename}, 0x7, value);
            ${devicename}->queue_command(set_payload_command);
```

Mosibi, 2024-10-21:
I do not have the app (Airwell has an issue at their side an say they are working on it) and also do not have a electrical heating element for DHW. So I am afraid that I am not able to test it right now

rysiulg, 2024-10-21:
So maybe wait to confirm by others before implemented it ;) -I left decision for you ;) if yes I prepare pull req. ;), No we wait ;)

rysiulg, 2024-10-21:
I've tested today this. After enabling in heatpump controller (unfortunetly this is only available on heatpump itself -not via modbus -for officially registers) than after switch on Forced Water Tank Heating and Forced Tank Buffer Heater working as expected, also when this is set by Mobile app smart home these values are set and don't revert after few seconds and on original controller there also icon showing that heater is enabled.
So only available testing is enable TBH / IBH on original controller service menu. Without that after setting value few moments later it is reverted. 
To disable it is needed to send value 2 to this register, but when register is read and function is off then read value is 0 ;) (sending 0 to turn off doesn't work).
So I thiunk we can make this update ;)

PS. for parameters 210 I see in york sm that bits 6, 8 and 14 could be read/only...  -but I'm not sure -so i think we wait to see if time shows true ;)

rysiulg, 2024-10-28:
Add pull request for setting Forced Electrical Heater for these Registers: 7, 8, 9 ? or we wait to future request from users?
For tests it must be enabled on wired controller that installation is equipped with electrical heater, after that on wired controller there icons for electrical heater are displayed, and can be set, also on mobile app this button Force Water Heating changes this value -Force Water Heating means "turn on electrical heater". 
it's a pity that the manufacturer did not provide for forced heating of the DHW, omitting the dT5s without electric heaters to heat the DHW water;)

Hendrik212, 2024-11-05:
This is an interesting topic as I'd like to control with Home Assistant when the internal backup heater is turned on and off. I played around with setting register 9 according to your example and seems to work with a few restrictions.

The first thing I noticed is I need to enable the "master" DHW switch although I don't have DHW by the heatpump. In some online manual I found the following passage: "If AUTO mode is on for room heating or cooling, the BACKUP HEATER cannot be selected
The BACKUP HEATER function is invalid if only ROOM HEATING MODE is enabled". I think that means you need to have both ROOM + DHW modes activated. Otherwise I cannot force backup heater on via the menu, I cannot change the menu entry from OFF to ON when DHW is disabled. I mean the following menu setting:
[image]

The second thing I noticed is that after forcing the IBH off by sending value 2, there is a cooldown of 5 minutes until you can force it on again. (You can send value 1 immediately and the register will save it, but the relais only clicks after 5 minutes)

Mosibi, 2024-11-05:
Yeah this is a bit of a difficult one. That logic that happens is, of course, implemented in the heatpump self and the goal of this ESPHome config is not to replace the logic in the heatpump, but to be an addition to it.

That is for example the main reason why I think that automations should not be implemented in this ESPHome config, but should be done for example in Home Assistant if a user wants that.

bkbartk, 2024-11-26:
the one for register 9 is interesting here.
personally I don't use IBH, it's disabled by the mentioned dip-switch in the screenshot above.
but AHS is enabled, and I think the same logic could/should be applied here.

However I want to disable AHS outside the boundaries of the ranch.
```
number.heatpump_ambient_temperature_trigger_for_ahs
number.heatpump_ambient_temperature_for_enabling_hydraulic_module_auxiliary_electric_heating_ibh
```
both have a range from -15 to 10. (I think both default to -5, this is a question in the yaml) 
But according to the manual my heatpump should be able to produce enough heat at -25, therefor -15 doesn't make any sense.

I actually only want to rely on, I think,
`number.heatpump_temperature_return_difference_for_enabling_the_hydraulic_module_auxiliary_ibh` 
which I prefer to set to 10

So it might would make sense to be able to turn it off, once turned on. I know the logic should be done by the controller, but those settings are actually limited.

rysiulg, 2024-11-27:
> the one for register 9 is interesting here. personally I don't use IBH, it's disabled by the mentioned dip-switch in the screenshot above. but AHS is enabled, and I think the same logic could/should [...]

OK.
I've checked in my manual for YORK Monoblock 9kW
Register 237: T4_AHS_o (umber.heatpump_ambient_temperature_trigger_for_ahs) default -5, min: -5 max: 30 -so only max is higher
Register 233: t4_ibh_on (heatpump_ambient_temperature_for_enabling_hydraulic_module_auxiliary_electric_heating_ibh) default, min, max same as Register 237.

So this min value could be manufacturer depended value...
I think -15 isn't bad value -heatpump efficiency drops a lot in that temp.
Personaly I don't use AHS nor IBH. My heatpump is directly connected to minibuffer (only 100l) and also i've connected Electric Heater in buffer, Gas boiler and coal-wood stove. And own logic for control. HeatPump is only on Heat or cool manual set.

closed 2025-04-13 by Mosibi