Mosibi/Midea-heat-pump-ESPHome #64: turn on fails

bkbartk, 2024-11-09:
This one might not be related to this project
I actually think for some reason the Midea controller is not accepting the request,
Still I post the issue here and I'm wondering if there would be a good solution for this.

I control My heatpump using automations
[image]
but as you can see 
at 06:00:01 my automation turns on the heatpump.
then 9 seconds later at 06:00:10 the device registers off,
I think information comes from the midea controller itself. but for some reason my turn on request was ignored.

Has someone else encountered this issue and can I would there be a workaround for this?
I ofc can send the turn on account multiple times but I doubt that would be a good solution.

Mosibi, 2024-11-09:
This is indeed done by the heatpump self. That specific entity is on my system completely in line with the thermostat signal. Thus if the thermostat turns on, this one is also turned on (by the heatpump) and goes to off, when the thermostat is turned off.

bkbartk, 2024-11-09:
the statuses indeed seem to be in line, no doubt about that.
still I'm wondering why my heatpump decided to turn off, there is no external thermostat or anything attached.
or maybe my heatpump never processed the turn on request correctly and just sends the current status on an interval.

I think I have to investigate some more

nikolamin, 2024-11-12:
on mhcv16wd2n8b when you adjust any parameter except the target temperature for dhw/heating/cooling the device is deactivating the dhw and heating/cooling. Check is your automation is setting any parameter that might deactivate the heat pump.

> or maybe my heatpump never processed the turn on request correctly and just sends the current status on an interval.
or maybe this, the entity in HA is updated optimistically and on the next refresh HA is updating the information about that the entity is actually off

bkbartk, 2024-11-12:
I only set target temp and on/off by automation,
So I expect the turn on signal was ignored somehow.

if this occurs more often I'm thinking about adding a check 1 minute after the update is applied and keep looping until the status is correct.

rysiulg, 2024-11-19:
@bkbartk check also that in (if remember) parameter 210 is enabled heating, and look for other sets -maybe is something that sends some value -every service parameter register (like delta temp, and much more -almost everyone which is on controller in service menu) disables power to heat/cool zone 1 zone2, DHW. 
If you manually turn on this also disable after while?

bkbartk, 2024-11-21:
no, nothing else has changed at that time, no one touched the controller at that moment. Everyone was a sleep,
and my automations do not touch other settings. So probably the turn on was ignored.
but it hasn't happened since, So I will close the topic for now.

closed 2024-11-21 by bkbartk