Mosibi/Midea-heat-pump-ESPHome #68: Errors not logged

CodrutTapu, 2024-11-25:
Hello! 

I have a Hyundai PDC and a few days ago i had a H7 error because of a voltage drop.
The error was logged in PDC but not in ESPHome.
 
I checked all these fields

[image]

Thanks

rysiulg, 2024-11-26:
OK. But this screen doesn't tell anything. After click on some entity than you can see history status:
[image]
If error isn't logged in HA than there could be other kind of problem type... ;)

CodrutTapu, 2024-11-27:
Sorry for not including more details. As you can see i had a voltage drop to 166V. I had the H7 error displayed on my heatpump screen but it was not logged in HA, as you can see.
Thanks
[image]

bguettner, 2024-11-27:
Hi, It looks like errors are actually logged, but once they are gone the status returns to "OK":
[image]
Using with York, ver. 4.1.2.

rysiulg, 2024-11-27:
If error will be temporary for short time and after that returns to ok it could not being read by controller -in section: modbus_controller->update_interval you have time specified to read from heatpump registry.
Some text valued sensors are visible only after zoom timeframe on chart -if they aren't permanent and changes faster.
Controller reads heatpump modbus and simply send data to homeassisatnt, where some data is recorded some displayed only (not all homeassistant records for history -it depends on config).

CodrutTapu, 2024-11-27:
@rysiulg the errors are stil displayed in the heatpump menu. Not on the main screen, but in the service screen. Updating update_interval to 1s for example can have any downside ?

bkbartk, 2024-11-27:
The other day I noticed some statuses not updating in HA either,
I can't tell if the statuses were OK in the esp web interface or not, But I assume the issue is related to ESPhome itself and not related to this implementation.

for me `_status_bit_1_defrosting` was `on` for over an hour, 
but `_active_state` was `Heating`(except for the first 2 minutes where is was defrosting) which is impossible if you check the lambda.
So HA wasn't aware of the status change of the defrosting bit.

If that can happen maybe some other status fails can also happen.

I noticed in the latest esphome release, 2 modbus fixes, maybe they are related.

rysiulg, 2024-11-30:
> @rysiulg the errors are stil displayed in the heatpump menu. Not on the main screen, but in the service screen. Updating update_interval to 1s for example can have any downside ?

I've tryed gets few seconds and I've get errors with modbus communicate. Get in mind that there is over 200 registers to read and interpret -so read evenm maked in whole range must be splitted by about 100 registers as limitation modbus/esp/library. So 1s is impossible with all sensors -maybe only few registers this will can handle ;)

bkbartk, 2024-11-30:
I just noticed another sensor not updating in ha. So I went to the esphome webinterface and there the sensor was ok. 
so it seems like a miscommunication between esphome and ha. 
maybe it’s the amount of sensors.

Mosibi, 2024-11-30:
@bkbartk I did not see that behavior yet. Can you tell more about the Esphome hardware you are using, which esp chip and amount of memory does it have?

if for example your esp chip is not able to keep up, you could change the update_interval to 30 seconds, or remove some registers that you do not need. If it happens often, then I am also curious if you can reproduce it with an older version. In version 4.0.0 I shifted the esp framework from arduino to esp-idf, maybe that is something you can play with.

Mosibi, 2024-11-30:
> I've tryed gets few seconds and I've get errors with modbus communicate. Get in mind that there is over 200 registers to read and interpret -so read evenm maked in whole range must be splitted by [...]

With this amount of entities, the default interval of 60 seconds should be okay. Be aware that the modbus component consolidates as much modbus registers in one call, so setting the interval on one sensor could influence more sensors than expected

bkbartk, 2024-11-30:
@Mosibi 
this is the behavior I meant
[image]
So far I only noticed this for binary sensors.

for hardware I use the module from SvenAr, not sure about the memory.
I use esp-idf, also before v4, arduino framework crashed sometimes for me if a value was set outside the valid boundries.

Today I notide this on v5, but earlier, before updating I noticed the same behavior on another sensor. But If I don't notice it doesn't mean it's not there.

There is one difference between the default and what I use.
For all my esp devices I use a base for wifi etc.... It's like this.
```
wifi:
  id: wifiId
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # power_save_mode: NONE
  power_save_mode: HIGH
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:
  
api:
  encryption:
    key: !secret api_encryption_key
  reboot_timeout: 1h

ota:
  platform: esphome
  password: !secret ota_password
  # safe_mode: true
  # reboot_timeout: 3min
  # num_attempts: 3
  
safe_mode:

web_server:
  port: 80
  version: 3
  local: true
  auth:
    username: admin
    password: !secret web_server_password

button:
  - platform: restart
    name: "${device_name} switch Restart"
    icon: "mdi:restart"
  - platform: safe_mode
    name: "${device_name} Restart (Safe Mode)"
    icon: "mdi:restart-alert"
```

maybe including safe_mode here, is just to much.
but the actual issue is in esphome, not in the yaml.

Mosibi, 2024-11-30:
The screenshots show a discrepancy within home assistant itself, the entity shows the binary sensor to be off, but the statistics show that it is on.

To me it seems that there is a issue in the (long term) statistics and not on the Esphome side. Is your home assistant able to keep up with the the load, is the underlying database okay?

bkbartk, 2024-11-30:
not within HA itself,
it's always ON
[image]
the right hand side is the webinterface v3 of esphome.

I haven't noticed such a behaviar on any of my other 40 esphome devices. My HA installation is able to keep up and no issues with the stats

the stats from my boiler(also esphome) show that my boiler was only turned on for 2 minutes, so I assume AHS was only enabled for 2 minutes(during defrost)

Mosibi, 2024-11-30:
Ah my mistake, I thought both screenshots where from home assistant itself. Okay so the strange behavior is only visible in HA, the Esphome device reports the right value.

I have not seen this before, you could enable debugging in Esphome to see if it is reported correctly to home assistant. Or maybe test with a config where only this sensor is presenting see if that changes something

VASTechAD, 2024-12-17:
Hello! I have encounter these discrepancies as well
On the left is the HA sensor logbook, on the right the live view of ESP controller.
It seems that at some point it was triggered, but never went back to real values.
Any option to solve this problem?
[image]

bkbartk, 2024-12-17:
I started a threat on the forum
https://community.home-assistant.io/t/esphome-entities-not-always-updating-in-homeassistant/802143
the issue seems to be esphome related, but I don't have a decent fix for is.

Mosibi, 2024-12-17:
> I started a threat on the forum https://community.home-assistant.io/t/esphome-entities-not-always-updating-in-homeassistant/802143 the issue seems to be esphome related, but I don't have a decent fix for is.

Thanks, I subscribed to that topic for updates. For now this issue can stay “open”, it can be closed when it it’s completely clear that is an Esphome issue.

bkbartk, 2025-01-12:
I added an issue for esphome
https://github.com/esphome/issues/issues/6664
and apparently there already is an MR that would fix the issue.
Over the last 2 days I havan't had the issue anymore. Still I do not know if this means 100% the issue is fixed.

Mosibi, 2025-01-12:
Good to see that it at least seems to work for now. Thanks for the update.

CodrutTapu, 2025-01-27:
Hello! Today i had the error logged for 1 minute. So the functionality is working properly. Most probably if the error is logged only for a few seconds is not logged properly. I will close this issue. Thanks

[image]

closed 2025-01-27 by CodrutTapu