Mosibi/Midea-heat-pump-ESPHome #69: Cannot get DHW T5 temperature

timomer, 2024-11-27:
Heat Pump: KAISAI ARCTIC KHC-10RY3-B

Hello, 

First thank you for this great work and the freedom you have given to owners of these heat pumps.

Im not sure if this may be me misunderstanding something or if my heat pump is reporting incorrect values. 
I am considering setting up a climate object to control the heat pump from HA fully. My first thought was to use the DHW sensor T5 as my temp sensor, but "DHW Tank Temperature T5s" always shows as 35. I can see [a comment here](https://github.com/Mosibi/Midea-heat-pump-ESPHome/issues/8#issuecomment-1893142473) about setting up a climate object, but again my readings do not seem correct:

"Flow temperature (T1)" - always 35, "Hydraulic Module Current 1" and 2 always 255.

I have asked the reseller for the modbus documentation and that's attached along with my config if helpful, I canont see anything wrong, I can only assume that I'm reading the values incorrectly or maybe there is a setting I need to activate to get 
the correct readings?

[Modbus KHA_KMK, KHC v2.pdf](https://github.com/user-attachments/files/17939781/Modbus.KHA_KMK.KHC.v2.pdf)
[Heatpump Controller.pdf](https://github.com/user-attachments/files/17939782/Heatpump.Controller.pdf)

Any suggestions? Thank you

Mosibi, 2024-11-27:
I would advise you to (let) test the water tank temperature sensor (T5) if it is working correctly. Hydraulic Module Current 1 and 2 are also 255 over here, so that is a sensor value that both our heat pumps have not implemented.

Regarding T1, which sensor do you mean exactly (do you know the modbus register?),. Could maybe related to “Parameter Setting 1 Supports T1 Sensor”, which is off on my heat pump

rysiulg, 2024-11-27:
Hi
T5S (sensor.heatpump_dhw_tank_temperature_t5s)  is DHW Temp Setting temp -the heat pump will goes with T5 temp to achieve T5S when heating DHW -so this value don't change -it is changed by user setting DHW temp which he want
To read water tank temp (DHW) you use sensor.heatpump_water_tank_temperature_t5
Unfortunatelly in config Register 4 was duplicated in sensor (Read only) and number (to be config set) -i've make pull request to remove from sensor ;)
Hydraulic Module Current 1 in my case is TBT sensor value -I have connected one temp sensor to DHW and second one (build by myself) to buffer. If you don't have this second sensor than you have 255 value in reading because you don't have enabled in parameters Setting 2 TW2 T1B Enabled -but when you enable this without sensor there will be error on display ;) -Register 210 bit1
"Flow temperature (T1)" -what sensor do you mean? 
In future is good to ask on sensor by Register number and if available bit number ;) -it will be easier to locate ;)
You don't have enabled water heating "Parameter Setting 1 EnableWater Heating" -so you cannot heat DHW and T5 values can be not real ;)
This parameter "Parameter Setting 1 HeatingAnd Cooling First Or Water First" forces DHW before Heat home ;)

PS. I don't use curves ;) -I manually set T1s remotelly ;)

Sensor "Parameter Setting 1 SupportsT1 Sensor" is wrongly named and should be "PUMPI silent mode, 1; valid, 0: invalid" -enables PUMP_I -pump in heatpump to work more silent ;) -In your modbus description is same as in my YORK ;)

"Restart with Factory DefaultSettings" means reset ESP config area -some sensor values can be saved to esp eprom so this only erases this area and after reading from heatpump it is restored from heatpump. This don't send anything to heatpump ;)

I've attached my view ;) -Register 4 is defined as sensor and number -I've just removed it "DHW Tank Temperature T5s" is to remove valid is "Set DHW Tank Temperature T5s" in my dump ;)

[york.pdf](https://github.com/user-attachments/files/17940805/york.pdf)

rysiulg, 2024-11-27:
> Parameter Setting 1 Supports T1 Sensor

/ @Mosibi I think it is wrong named as created new issue  for this -and removed as pull req. redefined REGISTER 4 as sensor and number, where sensor should be removed ;)

timomer, 2024-11-28:
Hi both, thank you for your fast replies. 

> In future is good to ask on sensor by Register number

yes sorry, this was an obvious oversight on my side.

My aim is to get the T5 temperature probe reading:
"sensor.heatpump_water_tank_temperature_t5" (Register: 115) for me is showing as 25 this is incorrect and has no updates.
> You don't have enabled water heating "Parameter Setting 1 EnableWater Heating" -so you cannot heat DHW and T5 values can be not real ;)

@rysiulg are you saying that the reason my T5 value is incorrect and not being updated is due to me not having the heat pump in DHW mode enabled?

I am assuming for me to take full control of the Heat Pump with a HA climate object that I should set the Heat Pump to DHW mode only (if possible) and use "sensor.heatpump_water_tank_temperature_t5" as my temperature sensor?

Does this sound reasonable? Im not using the heat pump for DHW, only heating, I just want as much control and logic processing to be done in HA to work with my heating system. 

> PS. I don't use curves ;) -I manually set T1s remotelly ;)

@rysiulg what is a "T1s"?  :see_no_evil:

Mosibi, 2024-11-28:
@timomer now I am confused, you state that you do not use the heat pump for DHW, only for heating, but at the same time you want to use the DHW tank sensor (T5) to read the temperature? 

T1S is the setpoint for the leaving water temperature (LWT), which is basically the temperature of the water running through your central heating system.

rysiulg, 2024-11-29:
No. I use my Heatpump to heat DHW and central heating my house ;). 
I just don't use curves and automation from heatpump -which isn't good for me ;)
T1s is temp to which heat pump heat for zone 1. 
Better is to describe your water connection.
for exampke main is:
Heat pump is connected to buffer directly withj attached TBT1 temp sensor. I set T1s to heat buffer to eg. 42 deg in day (i have hpe that sun give me some power) and by night at about 32 deg. I have connected electronic 3way valve to heat DHW. The heat from buffer is transported by independed Grundfoss Pumps -1 pump is for floor heat and second one is for radiand in my bathroom and recuperation heat-exchanger ;)
So when you don't have DHW heating enabled than T5 can show 25 deg as const value ;)
I choose to control from esphome as much as possible and in some ponts from ha -to avoid situation when HA is down that heatpump can be still working ok ;)
I put other control of PUMP_I and separate pump connected in line with heatpump to start this second pump in low temps. I've noticed last year that when other source is active for heating (in my case wood/coat heat source) than heatpump don't yurn on PUMP_I which makes me freeze my pipes in connection outside house. So I check how much time elapsed when PUMP_I don't run. If not run for some tme -i energize switch that enables for few secondes seconf pump to avoid freeze water in pipes ;)
If you don;t use Heatpump for DHW than T5 sensor is the sensor that you don't care of it ;)

timomer, 2024-11-30:
Hi @rysiulg 

You can read a full in-depth post about my setup and changes to it [here](https://www.greenbuildingadvisor.com/question/fixing-mistakes-of-the-past-thermal-store-and-fireplace-with-water-jacket)

But in summary, I have an 800l water tank top half heated by solar PV or fireplace back boiler and the bottom by a Heat Pump I installed a few weeks ago, DHW goes through a coil from bottom to top of the tank. The house was renovated about 5 years ago with the plan for a GSHP and at the time a large buffer was suggested, we went with Solar PV as it was cheaper and easier to have a large solar array (10KW) than worry about water freezing and running pipework for Solar water.

 Setup seems to be working well, with the UFH active the HP is set to provide 37c water, the water tank stays around 32-37 whats ok for showers and if we have a sunny day or put the fireplace on we have the top tank heat to 55c and that's ok for baths. 

Originally I never connected the T5 sensor but got an error on startup, hence I contacted it. Right now the HP is set to activate on "water flow temp", but I am confused about how it knows when to turn on as the water inlet temp is below target when the UFH is not on but the HP does not activate and when I turn on the UFH (from Home Assistant, has its own pump) about 5-10mins later the HP activates - this is the behaviour I want but I do not understand how it knows when to turn on, maybe there is a lower threshold for the "water flow temp" or I thought maybe it is looking at the T5 sensor reading and seeing the tank temp dropping?

From reading your comment I believe I need to use the TBT1 sensor as my target temperature and locate it in the bottom part of my water tank, but I also only wish for the HP to activate when I have heating demand so I need a switch I can toggle from HA - right now I am looking at using "switch.heatpump_water_flow_temperature_control_zone_1", what do you use or do you always have the HP active to keep the tank at temperature?

> I choose to control from esphome as much as possible and in some ponts from ha -to avoid situation when HA is down that heat pump can be still working ok ;)

Ideally, I would do the same, but the UFH and HP are controlled by different ESPHome devices and HA is the "middle man"  managing them.

timomer, 2024-12-03:
> but I do not understand how it knows when to turn on, maybe there is a lower threshold for the "water flow temp" or I thought maybe it is looking at the T5 sensor reading and seeing the tank temp dropping?

I believe I have answered this, looks like the water flow continues when the HP is inactive and only activates the HP when the water outlet and inlet differ. 

@rysiulg I would be interested to know what switch you are using to activate the heatpump?

rysiulg, 2024-12-03:
hifirst of all you have to get connected temp sensor T5 to water tank and 3way walve indicated in output as sv1 or sv2 (don’t remember)next you set T5s as setpoint temp for your dhwwhen t5 drops delta T5s (dT5s) default 5 deg in adjust range to 30deg than heat pump activate dhw heating (also in parameter 210 or 211 you need enable water heating)” and power on dhw in register 0)PozdrawiamRyszard GwizdakMARM.pl Sp. z o.o.

rysiulg, 2024-12-03:
with heatpump on standard you have one sensor which basically is used for heat. for dhw you must connect second sensor ntc about 47kOhm to water tank and enable it in parameters. i don’t buy it but make myself with connector ;)this is needed like in my case -my heatpump is parallel connected to buffer with Tbt sensor which gives me unlimited heat sources as buffer works as hydraulic „clutch”PozdrawiamRyszard GwizdakMARM.pl Sp. z o.o.

rysiulg, 2024-12-03:
so in short T5 is physical ntc temp sensorPozdrawiamRyszard GwizdakMARM.pl Sp. z o.o.

timomer, 2024-12-03:
Thanks @rysiulg ,

Im not worried about DHW as I get this from Sun or Fireplace (this may change later, but not now). 

I understand you use the T5 sensor for DHW, but what sensor / switch do you use for heating? Say your DHW is already at the target temperature, but your UFH is demanding heat, how do you trigger your HP to activate? 

> with heatpump on standard you have one sensor which basically is used for heat

Yes, this is what I want, one sensor for heat. What sensor are you using? I believe this can only be "Water flow desired temperature" or "Room desired temperature". Both of these do not work for me, ideally, I want heating to behave like DHW where a sensor in the tank triggers the HP. 

Right now I am using "Water flow desired temperature" and switch  "switch.heatpump_water_flow_temperature_control_zone_1" for my HA thermostat

rysiulg, 2024-12-03:
Hi

OK.

In my case I have connected sensor TBT -at buffer. My heatpump heats buffer directly to this temperature by water flow T1. So T1 sensor is outlet temperature which indicates Water Outlet Temperature from HeatPump.

So in my case is simple HeatPump heats until tbt reach target temp (In mossibi name is Hydraulic Module Current 1 -but in York and mide this is TBT1 temp). -Yes I preffer this -I can connect to buffer more than 1 heat source at same time. I control room temp from pump group and external reading of room temps ;) -in my case averange temp get from recuperation module.

If you don't use tbt (buffer) and DHW -than you don't need this attached sensor.

If your heatpump is directly connected to your radiants etc, even with buffer in return connected in serial than you need get temp from rooms -Room Temperature Ta. Heatpump is started/stopped by closing HT/COM wires from room thermostat. Heat pump don't know room temp and desired temp in direct connection case? or you can try use T1 sensor for heat but T1 isn't room temp but most often outlet temp after AHS/External Electric Heater temp acts as Water Flow Temp..

Ta is room temp, but as documentation says is internal temp. I think heatpump calculate it somehow. The logic in heatpump isn't good described and thats why my heatpump is working on manual mode. I turn on heat pump based on external sensors.

Ta-adj
This is a correction value for Ta which is inside the wired controller.

On attachment SM look at page 39 -There you have described HeatMode./
[MONOBLOK 201.55-M1.EN.CE (722) final_small.pdf](https://github.com/user-attachments/files/18000487/MONOBLOK.201.55-M1.EN.CE.722.final_small.pdf)

timomer, 2024-12-10:
Thanks for the feedback @rysiulg 

I have decided to do the following:

1. Setup HA Thermostat with a temperature sensor in the floor, and a switch that activates the UFH pump
2. Set a HA Automation triggered when UFH pump state changes and activates the HP with switch "switch.heatpump_water_flow_temperature_control_zone_1" 
3. HP manually has its temperature target set to 37 and is set to "Water flow desired temperature" 
4. Once HA Thermostat meets its desired temperature it turns the HP off, but I have added delays to be sure I am not turning the heatpump on / off often

This appears to be working fine. 

> So T1 sensor is outlet temperature which indicates Water Outlet Temperature from HeatPump.

If I understand correctly this is what I have set, in my documentaion its called  "Water flow desired temperature". 

My final question for you is: Are you turning the HP on / off when needed or do you just leave it to monitor the Water Outlet Temperature? If you are turning it on / off what switch are you using for this?

Many thanks

timomer, 2025-02-27:
An update to this: I have had the above setup for this heating season and happy to say it appears to be working perfectly. 

The HP is not cycling on/off, I'm using switch `switch.heatpump_water_flow_temperature_control_zone_1` to "power" the HP on and off based on UFH pump being activated.

Happy to close

closed 2025-02-27 by timomer