Mosibi/Midea-heat-pump-ESPHome #71: Wrong name for Register 210 bit 5 sensor

rysiulg, 2024-11-27:
Hi
For Register 210 bit 5 is:
```c++
  # Register: 210, Bit: 5, default: false, TODO: verify default
  # Midea, YORK: PUMPI silent mode, 1; valid, 0: invalid
  - platform: template
    name: "Parameter Setting 1 Supports T1 Sensor"
    id: "${devicename}_parameter_setting_1_supports_t1_sensor"
``` 
but this sensor name is "PUMPI silent mode, 1; valid, 0: invalid" so I think it should be change

Mosibi, 2024-11-28:
This is one of those where not all registers have the same meaning according the different manuals. I have to check this and will choose what the majority of the manuals state

bkbartk, 2024-11-29:
Mine, (Midea) has the same description as @rysiulg 
but I don't really know what it means or does.

rysiulg, 2024-11-30:
> Mine, (Midea) has the same description as @rysiulg but I don't really know what it means or does.

I think this defiunes that PUMP_I supports silent mode -so it just works in slower speed to make less noise ;) -you can after enabled -if PUMP_I supports it to set for silent or supersilent mode ;)

Stormhand, 2024-12-23:
On my Clivet monoblock its described as **Backup PUMPI silent pump function, 1:backup**

closed 2025-05-31 by Mosibi