Mosibi/Midea-heat-pump-ESPHome #73: Register #129,

bkbartk, 2024-12-12:
For register 129
I would like to change binary sensors to switches. (not sure if Midea accepts this)
is there anything I should be aware of?
and would it be smart to convert all bits at once, or only the one I need(14)?

Mosibi, 2024-12-12:
Register 129 delivers sensors. Those cannot be changed, from the documentation: "The following address table can only use 03H function code (read register)."

bkbartk, 2024-12-12:
that's unfortunate, Then I have to find another way

Mosibi, 2024-12-12:
What are you trying to do ?

rysiulg, 2024-12-12:
Reg 129 is only informational status of HP. Parameter 211 bit 14 you can choose eg to use M1M2 as AHS signal. 
But I think you can get problem in opentherm and CV control -I would do in that way that temp sensor I'll put on buffer (Heat pump 1 sensor and CV gas heater second one) and only control enable/disable CV to work. I think that gas burner don't stop because gas heater (CV) don't have required temp. You also have there temp reading and temp set values with specified delta -I have also connected gas source to my buffer -but not running. In my site wood heater is used for cold days ;)

Mosibi, 2024-12-13:
This is beyond my knowledge, my advice is to go your installer with this question or ask on a forum like https://renewableheatinghub.co.uk

closed 2024-12-13 by Mosibi