Mosibi/Midea-heat-pump-ESPHome #78: Installing on ESP8266?

ckszabi, 2025-01-25:
Is the code installable and runnable on a NodeMCU ESP8266?

Mosibi, 2025-01-25:
We only have run it on esp32 chips, but the config in the yaml is not esp32 specific, except of course for the board configuration. If you have the board at home, just try it, if you have to order a new board, then go for the esp32

closed 2025-01-25 by Mosibi

ckszabi, 2025-01-25:
Asking, because I actually have a ESP8266 board, added to ESPHome with "empty" config working fine, so enrollment is OK.
In the _heatpump.yaml_ I've changed the `esp32` section to meaningful `esp8266` content, uploaded to the board, but after that gets into reboot cycle. Might not have enough RAM...
Will get an ESP32 board. Thanks for quick reply.

Mosibi, 2025-01-25:
We use chips with 4MB memory with about 2MB used, the other 2 is meant for the ota update. You could remove the ota part in the config, which makes the complete memory available. Then updates are of course not possible using the ota method, an wired update is then needed.

ckszabi, 2025-01-25:
Removed OTA section, still getting in log "WARNING Memory allocation of 152 bytes failed at 40223D4F" and lot of stack trace...
so I feel that RAM is not enough. Anything else besides the OTA that can be removed, shrinked?

svenar-nl, 2025-01-25:
If you wish to strip it down to the bare minimum, you can also remove the `web_server:`, `wifi.ap:`, and `captive_portal:` sections and specify your WiFi credentials manually in the configuration.
Additionally, you can reduce the logging level from `INFO` to `WARN` to save memory.

ckszabi, 2025-01-25:
Removed `web_server`, `wifi.ap`, `captive_portal`
Still getting 
```
Unhandled C++ exception: OOM
```
and sensor allocator problems.
So, the 80KiB RAM of ESP8266 seems not enough for these many sensors. ESP32 has at least 320KiB RAM...

Mosibi, 2025-01-25:
You could strip the yaml to only the basic registers that you really need, but I do not expect that that will help a lot, because the most usage will be in the libs that are being loaded into memory.

ckszabi, 2025-01-25:
I've removed "half" of the sensors from yaml definition... still the same situation. Out Of Memory exceptions.
Hardware limits I feel, good to know from now on..
Thanks!