Mosibi/Midea-heat-pump-ESPHome #80: COP doesn't work (calculate)

radekoo92, 2025-02-10:
Hello everyone,

I used the configuration available here, but I noticed that two things that I would like to observe do not work for me. The first is COP, which has a constant value despite the entities that calculate it being active (power_output, electricity consumption).

The second thing is that the "defrosting" bit does not change during defrost. What could be the reason?

[image]

[image]

Mosibi, 2025-02-11:
The active_state value is derived from multiple sensors, one of them is the same defrost sensor (_status_bit_1_defrosting). Since _active_state is showing the right information, we know that _status_bit_1_defrosting does have the right information, but the update to home assistant is not always working well. Summary, using the active state to update the “defrost state” will not work. 

In the last version I added the _active_state_map sensor that is a number that reflects the active state. That one is correctly updated, so you could create a template sensor in home assistant that is true when _active_state_map is 4 and false when it is not 4. Personally I use _active_state_map to create a timeline view in Grafana with the active state as text.

But in the end, the issue must be fixed in esphome itself.

closed 2025-02-12 by radekoo92

rysiulg, 2025-02-13:
Yes COP isn't calculated reliably -it depends on heatpump measurments components which isn't super, but this is most easiest way to get COP and cheapest. Of course if you need more accurate data of COP than you must invest in elelctric meter (lets say about 100-200-USD) and Energy Meter -another few hundred of bucks for cheap meassurements and most often integrate into pipes... -but this isn't laboratory ;)

Mosibi, 2025-02-14:
Strangely enough the measurement of the heat pumps output is not that far of the reality, as I understood from people that measured with good energy meters in the piping.
So at home I use a Shelly (1PM) to measure the electricity usage and use that to calculate the COP. Then you also see how far the readings are off that are coming from the logic in heatpump.yaml.