Mosibi/Midea-heat-pump-ESPHome #81: Update framework to ESP-IDF for Midea Clivet R290 version.

mikidu57, 2025-02-24:
Hello,

First of all, I would like to thank Mosibi for his exceptional work.

I am currently using the code intended for the Midea Clivet R290 heat pump for a Ferroli Omnia 3.2 heat pump. I am experimenting with the code and adapting it with my limited knowledge to make it fully functional with this heat pump (some Modbus addresses are not equivalent, but 90% work).

However, I am experiencing a total crash of my ESP32 after a few days of operation. I believe this is due to a memory issue.

I have noticed that in the main version, the framework has been switched from Arduino to ESP-IDF. I know this improves memory management and multicore processing, but it also affects the use of lambdas. I don't think I have the knowledge to make these modifications without errors.

Would it be possible to update the Midea Clivet R290 branch accordingly?

Mosibi, 2025-03-09:
The Midea Clivet R290 branch was created to support somebody in the UK which had the first R290 model to see if the controller from svenar.nl was compatible with this specific model. We learned that it is compatible and kept this branch for historic reasons, but the branch is not maintained and sadly not on the same level anymore as the released versions for the non-R290's.

I just updated the branch with the esp-idf board type and also removed the `entity_prefix` from all entities so that is already in-line with the master branch.

mikidu57, 2025-03-12:
Hello Mosibi,

Ok, I understand better why the update hadn't been done. Thanks anyway for the modification to esp-idf.
I will continue working on adapting the code to my heat pump. I will run the test in two weeks because the heat pump is in my second home, 2000 km away. I prefer to be on-site to avoid the risk of losing the connection with my ESP32.

closed 2025-04-13 by Mosibi