Mosibi/Midea-heat-pump-ESPHome #82: How to turn ON/OFF heatpump (not an issue)

pacraf, 2025-03-22:
First thing first - Thank you dear developers for sharing this projects.

My pump is Midea branded as Kaisai Arctic 8kW Monoblock . 
It works from start with this ESPHome controller, I can change parameters, I see values in HA. Great!
I wonder how do you control your heatpumps. 
So the questions are:
1. I don't know how to perform complete On/Off by ESPHome. (I can do it only on original panel). Is it possible?

2. What are your automation "route" for heatpumps? (I mean here how your HA/ESPHome influences on heatpump, change on/off , emulate thermostat, change setpoint temperature) ?

Mosibi, 2025-03-22:
1. As far as I know, it is not possible to completely turn Off the heat pump. My own measurements also show that the power usage of an idle heat pump (both the internal (9 watt) and external unit (7 watt) ) is very low.
2. I use a custom weather compensation method: 
  a. A HA templated sensor (helper) calculates  the leaving water temp (LWT)
  b. An HA automation sets the LWT on the heat pump when the templated sensor has changed for 15 minutes. The 15 minutes is to prevent changing the LWT to often.
  c. A HA generic thermostat is used to decide if heating should be turned on or off, using a temp sensor in the living room

To summarize, my heating is on for 24 hours per day and if the temperature in the living room drops below the threshold, the generic thermostat enables the relay that is connected to the 'thermostat pins' of the heat pump. The LWT is configured continuously , maximal once per 15 minutes.

pacraf, 2025-03-22:
Thank you for answering.

Point 2 is clear. Summary as well.

Only to clarify point 1. I understand that Off state is not full power down
to zero watts. To achieve this I think only option is circuit breaker.
The question is whether it is possible to change state from Heating to Off
(and back too) - and I mean here Off is state that you pick from panel,
knowing that it will consume some watts of course.

If this is not possible, then is there known way to emulate a room
thermostat via modbus? You wrote that you use relay controlled by HA to
engage thermostat pins on heat pump. So there is no possibility to do this
just by modbus, right?

If above question answers are negative, I will also add a cable with relay
on espHome to emulate thermostat. Just wanted to avoid cable work, as I did
not added anything spare when installing , so laziness asks for modbus
support ;)

Mosibi, 2025-03-22:
Turning the heating to off is what I do with my relay on the thermostat pins. I do not know if that is the same as what happens when you turn it off on the panel.

As far as I know it is not possible to emulate the thermostat using modbus. But I am not an Midea modbus expert, I only have the docs they provide, so maybe there is a secret hidden somewhere :)

I am using a ESPHome based relay, which I got from https://shop.svenar.nl/ (not listed on there at the moment I see). If interested in his relay solution, you can sent him an email

pacraf, 2025-03-22:
Thank you for all the guidance.
Regarding espHome relay board, I have a drawer full of different boards,
here old good esp8266 will happily switch the relay…

I will close the issue when back to computer, thank you and have a good
weekend.

closed 2025-03-23 by pacraf

timomer, 2025-03-28:
@pacraf see last comment here: https://github.com/Mosibi/Midea-heat-pump-ESPHome/issues/69

pacraf, 2025-03-30:
@timomer thank you, you are right, mentioned  switch.heatpump_water_flow_temperature_control_zone_1 works as ON/OFF well.
to be clear - my heatpump works in simplest mode - only heating, only one zone, control mode without thermostat, just trying to keep constant outlet water temp.  I wanted to have option to  switch it on/off from HA - when not present home. And this seems to work.  no need for additional wire/board - laziness won! :) 
so below screen - just started heat pump from HA via ESPhome - great!
I only wonder - due to the name of this switch "water flow .... blahblah" - as this is monoblock, so better not to stop waterflow when temperature outside is below zero... but I suspect it's antifreeze option will kink in... anyway will test it, but not so soon. April almost here :)

[image]

timomer, 2025-03-31:
> as this is monoblock, so better not to stop waterflow when temperature outside is below zero... but I suspect it's antifreeze option will kink in...

I had the unit working all this winter and did not notice any issues. I agree that as long as the unit is powered, I would expect it has its own antifreeze behaviours that are automated. I would say its best to use `switch.heatpump_water_flow_temperature_control_zone_1` to control your heating and keep power to the unit.