Mosibi/Midea-heat-pump-ESPHome #101: Midea R32 + Atom S3 + RS485 Base

feeleep75, 2025-08-22:
Hi - did anyone have a success implementing this with AtomS3 Lite and RS485 base? I wa trying today for few hours and it always shows as "inactive" with no successful reading even a single parameter. I will appreciate example config file :)

Mosibi, 2025-08-24:
@feeleep75 For future reference, it would be nice if you can post the outcome in this issue and close it when you can.

feeleep75, 2025-08-24:
Sure - I am waiting for resistor now ;)

Mosibi, 2025-09-05:
The first error could be a serial message which is not complete received by your esp32. When that message does only sometimes occur, you can ignore it. 
The other 2 messages show that your heat pump returns some data that is not expected by the code and the a “unknown: <value>” is returned, see https://github.com/Mosibi/Midea-heat-pump-ESPHome/blob/b6e4c5fb0d91b947a30e09662fa9c958112c8398/source/heatpump-base.yaml#L181

What heat pump brand and model do you have?

feeleep75, 2025-09-06:
It is Midea R32

feeleep75, 2025-09-06:
I believe it is MHC-V22W/D2RN8

Mosibi, 2025-09-06:
That heatpump is often sold in combination with "remote controller" KJRH-120F-BMKO-E. I was not able to find a modbus register overview for your specific heat pump model, but all modbus register overviews I have and also the ones that I found in the last 30 minutes searching, have exact the same information. 

It's either

| Register address   |      Description      |  Remarks |
|----------|:-------------:|------:|
| 272 (PLC: 40273) |  EMISSION TYPE | Bit 12-Bit 15 Zone 2 C-emission, 0: FLH; 1: FCU; 2:RAD<br/>Bit 8-Bit 11 Zone 1 C-emission, 0: FLH; 1: FCU; 2:RAD<br/>Bit 4-Bit 7 Zone 2 H-emission, 0: FLH; 1: FCU; 2:RAD<br/>Bit 0-Bit 3 Zone 1 H-emission, 0: FLH; 1: FCU; 2:RAD|

or this one, in which case I do not have information about the exact mapping between the value and the emission type

| Register address   |      Description      |  Remarks |
|----------|:-------------:|------:|
| 272 (PLC: 40273) |  EMISSION TYPE |Bit12-15: The type of zone 2 end for cooling mode<br/>Bit8-11: The type of zone 1 end for cooling mode<br/>Bit4-7: The type of zone 2 end for heating mode<br/>Bit0-3: The type of zone 1 end for heating mode|

Some manuals (yes there are a lot of them, also from different vendors) have an overview like this

| Parameter  |      Description      |  Default | Min | Max | Unit |
|----------|:-------------:|------:| ------:|------:|------:|
| Zone 1 C-emission | The type of Zone 1 terminal for cooling mode: 0=FCU (fan coil unit), 1=RAD. (radiator), 2=FLH (floor heating) | 0| 0| 2 | / |
| Zone 2 C-emission| The type of Zone 2 terminal for cooling mode: 0=FCU (fan coil unit), 1=RAD. (radiator), 2=FLH (floor heating) | 0 | 0| 2| / |
|Zone 1 H-emission|The type of Zone 1 terminal for heating mode: 0=FCU (fan coil unit), 1=RAD. (radiator), 2=FLH (floor heating) | 1| 0| 2| / |
|Zone 2 H-emission|The type of Zone 2 terminal for heating mode: 0=FCU (fan coil unit), 1=RAD. (radiator), 2=FLH (floor heating)|2|0|2|/|

Can you like in the manuals (all of them) that you have for you heat pump and wired controller and see if you can find such information/table as above. I am interested in the mapping between emission type and the value, for example 1 is radiator and 2 is floor heating.

feeleep75, 2025-09-06:
Hmm - I think it can be completely different values for Zones emission because it changes very often in my system and values are just random like this:
```18:40:56
[E]
[select:018]
'Zone 1 End Heating Mode Emission Type': invalid state for publish_state(): Unknown: 9
18:40:56
[D]
[modbus_controller.select:014]
New select value 37945 from payload
18:40:56
[E]
[select:018]
'Zone 2 End Heating Mode Emission Type': invalid state for publish_state(): Unknown: 3
18:40:56
[D]
[modbus_controller.select:014]
New select value 37945 from payload
18:40:56
[E]
[select:018]
'Zone 1 End Cooling Mode Emission Type': invalid state for publish_state(): Unknown: 4
18:40:56
[D]
[modbus_controller.select:014]
New select value 37945 from payload
18:40:56
[E]
[select:018]
'Zone 2 End Cooling Mode Emission Type': invalid state for publish_state(): Unknown: 9```

Mosibi, 2025-09-06:
Can you enable line https://github.com/Mosibi/Midea-heat-pump-ESPHome/blob/b6e4c5fb0d91b947a30e09662fa9c958112c8398/source/heatpump-base.yaml#L164 
so that it prints the original value of register 272

Btw all entries with `[D]` are there because you enabled debug logging, but I assume you are aware of that.

Mosibi, 2025-09-07:
Very strange, it looks like that specific register returns data which alternates, like something like a rpm fan speed or actual energy usage. The modbus controller error is maybe related to what you see, but i cannot say that for 100%. Since you are getting data, it means that the CRC check was okay, which is done by the modbus controller, therefore it seems to be a valid value.

Is this the only register where you see this, or also on others?

feeleep75, 2025-09-07:
These are the only errors I've seen so far.

feeleep75, 2025-09-08:
So I disabled register 272 but "not enough data" error remains and there is another one:
```
[09:17:13][E][modbus_controller:788]: not enough data for value
[09:17:13][E][modbus_controller.select:034]: No option found for mapping 51775
```

Mosibi, 2025-09-08:
Are you able to relate this to a register?